going-digital/Talkie #9: must be const in order to be put into read-only section by means of

Dsk128, 2015-03-07:
Hi, I'm having problems when I tried to compile the examples codes. 

Build options changed, rebuilding all
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino/Arduino.h:28:0,
                 from Vocab_US_TI99.ino:19:
Vocab_US_TI99.ino:32:31: error: variable 'spARE' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
Vocab_US_TI99.ino:138:31: error: variable 'spHELLO' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
Vocab_US_TI99.ino:299:31: error: variable 'spTEXAS_INSTRUMENTS' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
Vocab_US_TI99.ino:304:31: error: variable 'spTHE' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
Vocab_US_TI99.ino:309:31: error: variable 'spTHESE' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
Vocab_US_TI99.ino:367:31: error: variable 'spWORDS' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
Error compiling.

Is this issue related to my Arduino version (1.60). I'm on OSX 10.9.5. I hope you can help me, thanks.

kd8bxp, 2015-03-07:
I am getting the same error on a linux box running 1.0.5 - but I uploaded the library to codebender.cc
and it compiled correctly.  So I figured it was my machine as I've had some issues with the linux IDE before
but if other people are seeing the same error, then it can't just be me!

Dsk128, 2015-03-07:
I think it's something related with Arduino 1.6. I installed the 1.0.6 version and it works!

sylvainf, 2015-04-05:
Hello,

It doesn't compile with modern compilers because data in flash (PROGMEM) must be declared  as const. But it's easy to fix : 1 - in examples add 'const' on each PROGMEM array declarations
2 - fix talkie.cpp line 91by adding a cast : setPtr((uint8_t*) addr);

assadollahi, 2015-04-06:
hi sylvainf,

i tried your fix, changed this in the "getting started" example:
"const uint8_t spDANGER[] PROGMEM = {0x2D, 0xBF, 0x21, 0x92, 0x59,..."
and tried to change in the talkie.cpp, but in line 91 there is no such line. 
thus, i changed the one in 97 from
"setPtr(addr);"
to
"setPtr((uint8_t*) addr);"
in the talkie::say routine

however, then i get this error:

error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
In file included from talkie_demo1.ino:14:0:
\Arduino\libraries\Talkie/talkie.h:15:8: error:   initializing argument 1 of 'void Talkie::say(uint8_t*)' [-fpermissive]
   void say(uint8_t* address);

don't know what to do now. any ideas?

sylvainf, 2015-04-08:
Ok, you must modify the prototype too : in talkie.h replace "void say(uint8_t* address);''   by "void say(const uint8_t* address);"