going-digital/Talkie #20: not working on arduino uno (help?)

ayerone, 2016-08-04:
I am trying to get this going on my uno.  I'm having two problems.

1) Errors regarding PROGMEM
  error: variable 'helloo' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 uint8_t helloo[] PROGMEM = {0x11, 0x12};

  I got around this (for now) by just removing the PROGMEM keyword so that the data just go into normal memory. but...

2) when I get it to "talk", it's not recognizable speech, but more like chirps and scratches. It sounds like pure glitches.

I would appreciate any help, or suggestions from someone who's gotten this to work in a recent arduino IDE.

thanks

ayerone, 2016-08-09:
My issue was the same as #9 

I got those fixes to work.  There are 4 changes.

  1) adding "const" before all PROGMEM declarations (i.e. "const uint8_t varNameSP[] PROGMEM...") in your sketch
  2) in talkie.h, changing "void say(uint8_t* address);" to "void say(const uint8_t* address);"
  3) in talkie.cpp, changing "void Talkie::say(uint8_t* addr)" to "void Talkie::say(const uint8_t* addr)"
  4) also in talkie.cpp, in the definition of Talkie::say(), changing "setPtr(addr);" to "setPtr((uint8_t*) addr);"

closed 2016-08-09 by ayerone

christurnerchris, 2019-01-06:
this worked for me 
https://forum.arduino.cc/index.php?topic=555808.msg4007395#msg4007395