going-digital/Talkie #22: Error including Talkie library in Arduino IDE 1.6.5

pratik7210, 2017-02-21:
I am unable to include Talkie library in Arduino IDE 1.6.5. It displays the following error-
Arduino: 1.6.5 (Windows 8), Board: "Arduino/Genuino Uno"

Specified folder/zip file does not contain a valid library.

I also tried copying the header file code, .cpp code and example code in one file but it syill won't work-
/*Arduino: 1.6.5 (Windows 8), Board: "Arduino/Genuino Uno"

_1_Getting_Started.ino:14:20: fatal error: talkie.h: No such file or directory
compilation terminated.
Error compiling.

Specified folder/zip file does not contain a valid library.

Please help me with this problem and tell me which version of IDE does the code work fine on??

q2dg, 2017-04-24:
It seems this project is pretty dead, unfortunately.

lokkiikkol, 2018-01-22:
you have to change the " " signs for < and >.