going-digital/Talkie #24: uint8_t sp[] PROGMEM (text to speech)

abymolina, 2018-03-06:
Hello, everyone

Im totally new to programming. We have a school project and I really need help to convert text to speech using the Talkie lib.

I use this link for reference
https://rimstar.org/science_electronics_projects/ardunio_speech_synthesizer_using_talkie_library.htm

I reall don't understand the "hex" equivalent code for "Ah, that's better." and "Turn off the light! I'm sleeping!"

Our project has an voice output. 
I need the word "Wall", "UP", "Turn Left", "Turn Right" , and "Overhead" in a male voice.

Thanks
Aby

deladriere, 2018-03-20:
Hi Aby
try this

wall:
0x28,0x87,0x6b,0x8d,0xd4,0x95,0x92,0x1c,0x8f,0x0d,0xd0,0x70,0xa3,0x08,0x56,0x6e,0x48,0x5c,0x89,0xc3,0xa5,0xdf,0x65,0x74,0x27,0x01,0xd3,0x71,0x97,0x30,0x93,0x24,0xcc,0x85,0x59,0xa2,0x6c,0x5d,0x30,0x1f,0x67,0x88,0xb3,0x71,0x11,0xa2,0xd9,0x21,0x8e,0xd9,0x25,0x4f,0xe6,0x96,0x28,0x26,0xa7,0x36,0xe9,0x5b,0xc6,0xb8,0x1c,0x82,0x68,0x6f,0x88,0x6b,0x74,0x88,0xa3,0xfd,0x61,0xcc,0x51,0x2e,0x8f,0xe6,0x87,0xa9,0x4e,0xb9,0x36,0xea,0x5f,0xc6,0x3c,0x65,0xe6,0xa8,0x7f,0x90,0x72,0x32,0x6b,0x7d,0x7a,0x87,0xec,0xd2,0x24,0x70,0xfb,0x08,0xb3,0x43,0xa3,0xc0,0xde,0x46,0xee,0x16,0x03,0x00,0x00,0x00,0x00,0x00,0x00,0x0f
up:
0x25,0xcf,0xa6,0x27,0x24,0x27,0x36,0x2d,0x85,0x29,0xd7,0x1a,0x5d,0xb4,0x98,0x26,0xcd,0x72,0x54,0x68,0x43,0xe8,0x54,0xb5,0x49,0xa8,0xf5,0x79,0x9c,0x2d,0x22,0x01,0x00,0x69,0x4f,0x6b,0x0f,0x97,0x38,0x00,0x00,0x00,0x00,0x00,0x00,0x0f
tl:
0x26,0x9e,0x23,0xd2,0x23,0xe4,0xb6,0x78,0x8d,0x54,0x35,0x4f,0x5a,0xc8,0x51,0x2a,0x5c,0x35,0x69,0x88,0x7b,0x8c,0x70,0x37,0x67,0x29,0x48,0x59,0xd3,0x8a,0x6a,0xb5,0x20,0x65,0x4d,0x2b,0x9e,0xdd,0x82,0x9c,0x35,0xb5,0x65,0x51,0x0b,0x72,0xd6,0x94,0xd6,0x59,0x2d,0xc8,0x41,0x52,0x46,0x17,0xa5,0x20,0x1b,0x2d,0xae,0x74,0xe8,0x82,0x20,0x74,0x2d,0xd5,0x91,0x0b,0xbc,0xf4,0xb5,0x64,0xc7,0x21,0xf0,0x2a,0xda,0xdc,0x5c,0x05,0xc6,0xe8,0x2e,0x53,0x4f,0x1d,0x18,0xab,0xb6,0x54,0xbc,0x4e,0x89,0x93,0x98,0x16,0xca,0xba,0x2d,0xce,0x7a,0x4b,0xb8,0x27,0xaf,0xb8,0xe8,0x6e,0x95,0xae,0xbc,0xf2,0xaa,0xb3,0x4d,0xbb,0xf2,0xc8,0x9b,0x8e,0x32,0xab,0xd0,0x25,0x1f,0x56,0x4a,0xa5,0x42,0x97,0xa0,0x1a,0x2f,0xf5,0x36,0x12,0xe2,0xa2,0xa2,0xdd,0x33,0x48,0xca,0x8b,0x8e,0x09,0xaf,0xc2,0xc9,0x2d,0xca,0x27,0xdd,0x4b,0x3b,0xb7,0x29,0xab,0x74,0x9d,0x64,0xf2,0x19,0x2b,0x22,0xa2,0x6e,0xb8,0x76,0xee,0xee,0x88,0x24,0x2e,0x5f,0x79,0xba,0x2b,0x9a,0xb8,0x76,0xb7,0xad,0xe9,0xb0,0xac,0xda,0xb9,0x26,0x9c,0x48,0x0c,0xb0,0xf6,0xc6,0xad,0x34,0x8e,0xed,0xf2,0x1f,0x3b,0x4c,0x29,0xb2,0xca,0x6f,0xdc,0x2c,0x03,0x29,0xe4,0x7f,0x71,0xab,0x8a,0xe3,0x02,0x00,0x00,0x00,0x00,0x3c
tr:
0xae,0xdd,0x21,0xc2,0x22,0xe2,0xb4,0x78,0xa7,0x54,0x8d,0x48,0x52,0xf2,0x95,0x2b,0xd4,0x2c,0x69,0xc8,0x5b,0x4e,0xb7,0x88,0x26,0x25,0x88,0x45,0xca,0x86,0x53,0x17,0x28,0x15,0x6e,0x1d,0x6e,0x55,0xa0,0xd4,0xa8,0x6c,0x34,0x73,0x09,0x52,0xc7,0xf4,0xf1,0x2c,0x21,0xc8,0x03,0xcd,0xa6,0x94,0x98,0xd8,0x06,0x4f,0x99,0xa4,0xac,0x62,0xed,0x73,0xb0,0x43,0xb6,0x8a,0x4d,0x8e,0xc4,0x0d,0xc5,0x0e,0xf1,0x43,0x4c,0xd7,0xdd,0x04,0xc4,0x2f,0x36,0x1f,0xf3,0x92,0x90,0xb8,0xd8,0xad,0x25,0x6b,0x89,0xd3,0xe4,0xb0,0x16,0x2f,0x25,0x4e,0x5d,0x4a,0x4b,0x52,0xb5,0x38,0xe7,0x28,0x2a,0x99,0x33,0xf2,0xe2,0x62,0x28,0x7d,0x4a,0xcb,0xab,0xf1,0xe6,0xf0,0xd1,0x25,0xa8,0x5a,0xdb,0xd2,0x0b,0x97,0xb8,0x28,0xab,0xa8,0x30,0x5d,0xf2,0x6a,0x39,0x63,0x2b,0x4c,0x80,0xab,0xa6,0xe8,0xed,0xc2,0x01,0xee,0x86,0xa5,0xae,0x06,0x06,0xac,0x1b,0x92,0xda,0x2e,0xe8,0xf0,0x26,0xc9,0x76,0x3b,0x90,0xca,0x2b,0x15,0xdf,0xcd,0x40,0xa8,0x4d,0x58,0x72,0x46,0x05,0x31,0x6c,0x9b,0x10,0xcd,0xaa,0x15,0xb0,0xa5,0x42,0xa4,0xbb,0xb6,0xc2,0x8e,0xab,0x84,0x88,0x35,0x00,0x00,0x00,0x00,0x00,0xf0
overhead:
0x09,0x0e,0x6e,0x17,0x24,0xef,0x24,0xd8,0xdb,0x59,0x94,0xbc,0x1d,0x20,0x9b,0x66,0xd0,0xeb,0xb2,0x09,0x4c,0xab,0xc2,0xc8,0xc9,0x22,0x08,0xdc,0xab,0x54,0xa4,0x06,0xb7,0x09,0x55,0x77,0xb5,0x1d,0x70,0x5b,0xc3,0x34,0x22,0x49,0x20,0xfc,0xa0,0x4c,0xf7,0x5a,0x89,0x0c,0x9d,0xb2,0x52,0x66,0xa4,0x20,0x54,0xea,0x2a,0x9e,0x14,0x82,0x58,0xa8,0x23,0x64,0x92,0x8b,0xa3,0xe7,0xf0,0x82,0xd8,0x29,0x2e,0x5a,0x42,0x8c,0x13,0xa7,0x38,0x73,0x8d,0x34,0xa3,0xd2,0xe2,0xa2,0x2d,0x33,0x5c,0xf2,0x8a,0x9b,0xd3,0x88,0x2a,0xcb,0x2b,0x6f,0x4e,0xd3,0x6b,0x2c,0x8f,0xbc,0x59,0x4d,0xaf,0xb1,0x34,0x82,0x6a,0xb4,0xbc,0x47,0x54,0x6b,0xab,0xb6,0xcc,0x9c,0x40,0x2d,0xa8,0xca,0xb2,0xb2,0x03,0xb5,0xb8,0x69,0xad,0xc8,0x09,0xd5,0xe2,0xaa,0x3d,0xa2,0xd6,0x54,0x89,0xab,0xd4,0x8c,0x1c,0x53,0x26,0xcf,0x42,0xca,0x6a,0x44,0x03,0xc9,0x23,0x8b,0x6b,0x53,0x23,0x26,0xef,0x02,0xbb,0xa2,0x0c,0xab,0x79,0x28,0x8e,0xf0,0x16,0x4c,0xda,0xca,0x29,0xcd,0x83,0x2c,0x00,0x00,0x00,0x00,0x00,0x3c

abymolina, 2018-06-08:
Hi! thank you soo much for the reply