going-digital/Talkie #30: What should I add to this code to increase the volume for the sound in the speaker..??

tanmayi06, 2018-09-12:
#include "talkie.h"

Talkie voice;
const int8_t speech[] PROGMEM={0x00,0x00,0x00,0x08,0xd8,0x39,0x54,0x00,0xa7,0xa6,0x32,0xe0,0x04,0x46,0x55,0x28,0xde,0x53,0xaa,0xa9,0x42,0xe9,0x69,0x95,0xbb,0x4f,0x4c,0xa5,0xe7,0x51,0x15,0x5e,0x31,0x94,0x91,0x49,0xee,0xa8,0xc5,0x50,0x45,0x2a,0xb9,0x13,0x02,0x43,0xad,0xb9,0xf7,0x9d,0x0b,0x76,0x35,0x17,0x31,0xb7,0x46,0xd8,0x35,0x96,0x69,0xdf,0xb8,0x60,0xd3,0x54,0xc0,0x31,0x97,0xa2,0x14,0xf0,0x27,0xa3,0x02,0xfe,0xbb,0x52,0xc0,0x3f,0x3b,0x0a,0x38,0x79,0x42,0x00,0xbb,0xb4,0x2b,0x60,0x06,0x43,0xd7,0x65,0x4e,0xde,0xe9,0xa5,0x52,0x1f,0x94,0x56,0x58,0x84,0x2c,0x7d,0x34,0x9e,0xe6,0x15,0xb0,0x0d,0xd1,0x87,0x8b,0x77,0xc8,0xd4,0x87,0x58,0x4e,0x5e,0x11,0x52,0x1f,0x7c,0x17,0x78,0x95,0x0c,0xbd,0x77,0xdd,0x60,0x55,0xc1,0x75,0xce,0x6c,0x13,0xd7,0x04,0xd7,0x05,0x91,0xc3,0xec,0x33,0x44,0x67,0x79,0x0c,0x53,0x34,0x02,0x04,0xec,0xe4,0xcc,0x80,0x9d,0x53,0x10,0x30,0x05,0x13,0x69,0x3a,0x26,0xce,0xb4,0xd4,0xaa,0xc9,0x48,0xa3,0xc2,0x1a,0xa4,0x2e,0xb0,0xa8,0x76,0xab,0xe0,0x3a,0xab,0xb2,0xda,0x24,0x02,0x1a,0x89,0x9e,0x74,0x33,0x07,0x80,0x80,0x95,0xcd,0x19,0x30,0x9d,0x85,0x00,0xba,0xf6,0x14,0x25,0x0f,0x9d,0x6a,0x5e,0xc1,0x95,0x22,0x75,0x48,0x46,0x88,0x54,0xba,0x10,0xa1,0xd9,0x25,0x53,0xe9,0x62,0xb8,0x77,0x95,0x0c,0xa5,0x4e,0x69,0x39,0x16,0x31,0x94,0xaa,0xba,0xe5,0x48,0x2c,0x57,0xda,0x26,0xda,0x23,0x0d,0x5d,0xe5,0x1b,0x7b,0x8f,0x34,0x72,0x75,0x28,0x14,0xd5,0xd2,0xd0,0x34,0xa9,0xa1,0xe5,0x48,0x43,0xd5,0xa6,0x8a,0x96,0x23,0x09,0x55,0x97,0x2a,0xa9,0xad,0xc4,0x52,0x5d,0x0e,0xaa,0xbc,0xd2,0x58,0xf4,0xd5,0xa9,0x41,0x46,0x1c,0xf2,0x0f,0xc3,0x2a,0xc2,0x8d,0xc9,0x3f,0x93,0x99,0x12,0x26,0x46,0xc0,0xca,0xa2,0x00,0x00,0x00,0x0f};
const int ProxSensor=2;
int inputVal=0;

void setup() {
  pinMode(13,OUTPUT);
  pinMode(7,OUTPUT);
  pinMode(8,OUTPUT);
  pinMode(ProxSensor,INPUT);
  Serial.begin(9600);
  // put your setup code here, to run once:
}

void loop() {
  if(digitalRead(ProxSensor)==HIGH)
  {
    digitalWrite(13,HIGH);
  }
  else
  {
    digitalWrite(13,LOW);
  }
  inputVal=digitalRead(ProxSensor);
  Serial.println(inputVal);
  if(inputVal==0)
  {
    digitalWrite(7,HIGH);
    //digitalWrite(8,HIGH);
   // voice.say(speech);
    //delay(3500);
    //exit();
  }
  else
  {
    digitalWrite(7,LOW);
    //digitalWrite(8,LOW);
  }
  if(inputVal==0)
  {
    voice.say(speech);
    delay(2500);
    //exit();
  }
  
  // put your main code here, to run repeatedly:

}

deladriere, 2018-09-12:
Hi, 
nice sound ! where does it comes from?

jremington, 2018-09-12:
To increase the speaker volume, you need an amplifier. Tell us about your output circuit.

deladriere, 2018-09-12:
@jremington  or check this out too : https://github.com/going-digital/Talkie/pull/8/commits/cba63c074818afb0ac1c16a6aba25e73b5a95744

jremington, 2018-09-12:
I tried the programmatic approach mentioned above, and found that the output volume of the code can't be increased very much without distorting the output. A suitable audio amplifier on the output is definitely the way to go. 

You may need a low pass filter on the output to prevent the high frequency PWM from getting to the amplifier.

deladriere, 2018-09-12:
@jremington Yep agreed! I had also experienced distortion with this piece of code, so of course a amp circuit is the solution

tanmayi06, 2018-09-13:
>           To increase the speaker volume, you need an amplifier. Tell us about your output circuit.

In the output I just connected a digital ir sensor to the Arduino,2 LED,s to the Arduino and 2 pins to the speaker..(one is to the gnd nd the other to digital pin 3..??)

Thank q for ur reply..:)

jremington, 2018-09-13:
You will damage the Arduino, connecting a pin to a speaker. Use an amplifier of some sort.

ArminJo, 2020-06-05:
Use the Arduino Talkie library and connect the Speaker (> 32Ohm) betwen the non inverted / 3 and Inverted / 11output.