going-digital/Talkie #34: How to Convert word into Uint8_t sp() PROGMEM (TEXT TO SPEECH)

faith0319demo, 2019-07-13:
HI Guys,
 I am totally new to programmer. I have a project that related to " IR Sensor" and i really want to convert some word into Talkie Library.

My Project is like when you triggered the sensor then it will receive the signal and the sound will come out of speaker.

I have watched some examples from Youtube but I still cant understand those hex equivalent code .

I need the word "Welcome", "To", "The", "EE", "Cafe" for my project's Output

Thanks,
Faith

abbas1234566, 2020-02-20:
welcome to mechanical And automation department sir please help as card convert

ktychab, 2020-02-20:
> HI Guys,
> I am totally new to programmer. I have a project that related to " IR Sensor" and i really want to convert some word into Talkie Library.
>
> My Project is like when you triggered the sensor [...]

If you look there is a small converter program. I'll look and attach it here.